CalPolyRobotics/IGVC-OlympusFirmware #63: Confirm that Apollo firmware is working as expected

evanash16, 2019-10-12:
Manually test the firmware on Apollo (with while loops and such) to make sure that "set_headlights", "set_turn_signals", etc. are working as expected.

evanash16, 2019-10-12:
The headlights and turn signals were on the wrong GPIO ports, flipping them around seemed to put the turn signals in the right order according to the schematic.

evanash16, 2019-10-13:
The hardware expects the headlight signal on PA0, but the timer is configured to output PWM on PA2. Since PA1 and PA2 need to be the left and right turn signals, we need to find a way to output a PWM signal on PA0, or add a jumper to compensate.

elashley, 2019-11-02:
I got the headlights and turn signals working. Now, I need to get MISC_LIGHTS_1, MISC_LIGHTS_5, and MISC_LIGHTS_6 working.

evanash16, 2019-11-09:
I just created a pull request for my Apollo changes. https://github.com/CalPolyRobotics/IGVC-OlympusFirmware/pull/66

evanash16, 2020-01-11:
The [fix](https://github.com/CalPolyRobotics/IGVC-OlympusFirmware/commit/bea648062ffe7a864106815f858df6f0fe711dce) was tested with an oscilloscope and has been merged into I46-R3. We are good to close this out.

closed 2020-01-11 by evanash16